robotmcp/ros-mcp-server #63: Add launch file example to start MCP server/LLM client at robot bringup

stex2005, 2025-09-10:
Currently, the workflow requires installing the MCP server, then manually launching the LLM client along with the ROS nodes and rosbridge. Having a ROS launch file that ties the MCP server into the bringup process would simplify this workflow and reduce the number of manual steps.

Suggestion:

- Add a minimal launch example that brings up the LLM client with a sample robot stack (e.g., rosbridge + demo nodes)
- Document how to adapt the launch file for different robot systems and LLM apps

mhubii, 2025-09-13:
Maybe on a related note, would it somehow be possible to inject some sort of system prompt on launch?

stex2005, 2025-09-16:
I created some scripts to orchestrate http-ros-mcp-server/tunneling(ngrok)/basic ros nodes in one pull request.

launch.py is more effective for ROS-native scripts as the nodes are managed by ROS. To orchestrate non-ROS nodes I think it's best to use plain .sh launchers.

closed 2025-09-17 by stex2005