robotmcp/ros-mcp-server #94: Reorganize LLM integration examples into unified folder structure

stex2005, 2025-09-19:
Currently, we have different examples for different LLMs scattered across the repo. It would be a good time to reorganize them for consistency and discoverability.

So far we have integration examples for:
- Claude (stdio)
- ChatGPT (only http)
- Gemini (not tested)
- Cursor (both http and stdio)

**Proposal**

Reorganize into a single top-level folder, for example:

```
0_llm_integration/   (or other suggested name)
├── 1_claude
├── 2_chatgpt
├── 3_cursor
├── 4_etc
```

This will make it easier to:
- Add new LLM integrations without cluttering the root examples folder
- Compare approaches across providers (stdio vs http, etc.)
- Encourage contributions for other LLMs

rjohn-v, 2025-09-19:
I think both. Organize the examples by type in the folder.
Update the installation instructions linked to the installation readme.

stex2005, 2025-09-19:
Got it, working on it.

stex2005, 2025-10-05:
I’d like to add a new `client` folder, where each LLM client will have its own README with setup and usage details.

This folder will also include an on-prem implementation of a gemini client (which I recently tested and working well), as discussed in [#62](https://github.com/robotmcp/ros-mcp-server/issues/94#62)
.

stex2005, 2025-12-17:
Please use `robotmcp/demos-ros-mcp-server` for future integration demos/examples.